har-in-air/ESP32C3_BLUETOOTH_AUDIO_VARIO #2: SPI Problem

Levardar, 2023-03-07:
Hello.

I am trying to run my new miniVario based on your nice project.
I have problems connecting to MS5611 and MPU9250(CJMCU-10DOF). As I start the device I have this errors:

`Audio indication of battery voltage

Battery voltage = 4.05V
E (5945) ledc: ledc_get_duty(739): LEDC is not initialized

Check communication with MS5611
crcPROM = 0xF, crc calculated = 0x0
Bad CRC read from MS5611 calibration PROM
MS5611 OK

Check communication with MPU9250
MPU9250 ID = 00, expected 0x71
Error reading Mpu9250 WHO_AM_I register
MPU9250 OK`

I commented out the powerOff() option to see what's happening after checking the connection with them, that's way continues to MS5611 OK and MPU9250 OK.
I think there is a problem with the SPI connection. I checked all the connection but everything looks fine.
I am using ESP32-C3-WROOM-02 with the same connection as you have in REV B
Platform.io is set up to use this version

`; HW REV B
build_flags =
	-DARDUINO_USB_MODE=1
	-DHW_REV_B
;	-DARDUINO_USB_CDC_ON_BOOT

; HW REV A
;build_flags =
;	-DHW_REV_A`

Maybe you could give me some advice to debug this problem when you have time.

Thank you,
Razvan

har-in-air, 2023-03-08:
Try comparing the schematic of the wroom module versus the c3fn4, also the imu sensor board schematic  if you used a different  one. Could try putting  10k pullup resistors on the cs lines for the ms5611 and mpu9250. Maybe  obvious  but check the power supply for the sensors

Levardar, 2023-03-10:
I tried with the pullups but without success and then I use a oscilloscope to test the CSB(without 10K pullups). On start is going HIGH but never goes LOW.

har-in-air, 2023-03-10:
Try printing  out what pinNCS and pinCSB are in spi_init()

Levardar, 2023-03-10:
Spi_init debug:
MS5611 CS pin status: 0
MPU CS pin status: 1

har-in-air, 2023-03-11:
I would first check the WROOM module schematic against the C3FN4 schematic for differences.
I haven't had trouble with a hand wired prototype, or with the assembled pcb, and it was independentlyassembled and tested in France by my partner on the project.
You could try setting the CS pins to output and drive to 1 in the code before calling spi_init()

devenergize, 2023-03-12:
I found that the cjmcu-117 was the problem. I change it and now it's working. I'm just waiting to fly with it and test it. Thank you for your support and for the cool project you've done!

Levardar, 2023-03-12:
I found the problem, it was the cjmcu-117. I change it and now it's working. I am just waiting to fly with it and test it. Thank you for your support and for this cool project.

closed 2023-03-12 by Levardar