har-in-air/ESP32C3_BLUETOOTH_AUDIO_VARIO #6: Compile results in several 'USBSerial' was not declared in this scope

davinderc, 2025-03-03:
I'm trying to compile for the seeed_xiao_esp32c3 (I have an ESP32-C3 Super Mini) on a Mac using PlatformIO and I cannot seem to figure out why I get errors related to `USBSerial` not being declared in `config.h`. Below is my `platformio.ini` and aside from that everything else is as checked out from main. Some of the compiler errors are below.

https://github.com/har-in-air/ESP32C3_BLUETOOTH_AUDIO_VARIO/blob/552cd865ce0543e66bbb5320b5269188a9265681/src/config.h#L38

```[env:esp32c3]
platform = espressif32 @ 6.9.0
framework = arduino
board = seeed_xiao_esp32c3
board_build.flash_mode = qio
board_build.partitions = partitions.csv
board_build.filesystem = littlefs
board_build.f_cpu = 160000000L
board_build.f_flash = 80000000L
build_flags = 
	-DARDUINO_USB_MODE=1
	-DHW_REV_B
lib_deps = 
	h2zero/NimBLE-Arduino@^1.4.0
	AsyncElegantOTA
	AsyncTCP
	https://github.com/har-in-air/ESPAsyncWebServer.git
upload_speed = 921600
monitor_speed = 115200
build_type = debug
monitor_filters = esp32_exception_decoder
upload_port = /dev/cu.usbmodem1101
monitor_port = /dev/tty.usbmodem1101
```

```
Building in debug mode
Compiling .pio/build/esp32c3/src/adc.cpp.o
Compiling .pio/build/esp32c3/src/audio.cpp.o
Compiling .pio/build/esp32c3/src/ble_uart.cpp.o
Compiling .pio/build/esp32c3/src/imu.cpp.o
Compiling .pio/build/esp32c3/src/kalmanfilter4d.cpp.o
Compiling .pio/build/esp32c3/src/main.cpp.o
Compiling .pio/build/esp32c3/src/mpu9250.cpp.o
Compiling .pio/build/esp32c3/src/ms5611.cpp.o
Compiling .pio/build/esp32c3/src/nvd.cpp.o
Compiling .pio/build/esp32c3/src/ringbuf.cpp.o
In file included from src/mpu9250.cpp:2:
src/mpu9250.cpp: In member function 'int MPU9250::check_id()':
src/config.h:38:16: error: 'USBSerial' was not declared in this scope
 #define Serial USBSerial
                ^~~~~~~~~
src/config.h:152:26: note: in expansion of macro 'Serial'
  #define dbg_printf(x)  {Serial.printf x;}
                          ^~~~~~
src/mpu9250.cpp:33:2: note: in expansion of macro 'dbg_printf'
  dbg_printf(("MPU9250 ID = %02X, expected 0x71\n", id));
  ^~~~~~~~~~
src/config.h:38:16: note: suggested alternative: 'Serial'
 #define Serial USBSerial
                ^~~~~~~~~
src/config.h:152:26: note: in expansion of macro 'Serial'
  #define dbg_printf(x)  {Serial.printf x;}
                          ^~~~~~
src/mpu9250.cpp:33:2: note: in expansion of macro 'dbg_printf'
  dbg_printf(("MPU9250 ID = %02X, expected 0x71\n", id));
  ^~~~~~~~~~
src/mpu9250.cpp: In member function 'void MPU9250::get_calib_params(CALIB_PARAMS_t&)':
src/config.h:38:16: error: 'USBSerial' was not declared in this scope
 #define Serial USBSerial
                ^~~~~~~~~
src/config.h:151:26: note: in expansion of macro 'Serial'
  #define dbg_println(x) {Serial.println x;}
                          ^~~~~~
src/mpu9250.cpp:55:2: note: in expansion of macro 'dbg_println'
  dbg_println(("Calibration parameters from NVD"));
  ^~~~~~~~~~~
In file included from src/nvd.cpp:3:
src/nvd.cpp: In function 'void nvd_calib_load(CALIB_PARAMS_t&)':
src/config.h:38:16: error: 'USBSerial' was not declared in this scope
 #define Serial USBSerial
                ^~~~~~~~~
src/config.h:151:26: note: in expansion of macro 'Serial'
  #define dbg_println(x) {Serial.println x;}
                          ^~~~~~
src/nvd.cpp:17:3: note: in expansion of macro 'dbg_println'
   dbg_println(("Preferences 'calib' namespace not found, creating with defaults"));
   ^~~~~~~~~~~
src/config.h:38:16: note: suggested alternative: 'Serial'
 #define Serial USBSerial
                ^~~~~~~~~
src/config.h:151:26: note: in expansion of macro 'Serial'
  #define dbg_println(x) {Serial.println x;}
                          ^~~~~~
src/mpu9250.cpp:55:2: note: in expansion of macro 'dbg_println'
  dbg_println(("Calibration parameters from NVD"));
  ^~~~~~~~~~~
src/config.h:38:16: error: 'USBSerial' was not declared in this scope
 #define Serial USBSerial
                ^~~~~~~~~
src/config.h:152:26: note: in expansion of macro 'Serial'
  #define dbg_printf(x)  {Serial.printf x;}
                          ^~~~~~
src/mpu9250.cpp:56:2: note: in expansion of macro 'dbg_printf'
  dbg_printf(("axBias %d, ayBias %d, azBias %d\n",axBias, ayBias, azBias));
  ^~~~~~~~~~
In file included from src/ms5611.cpp:2:
src/ms5611.cpp: In member function 'void MS5611::averaged_sample(int)':
src/config.h:38:16: error: 'USBSerial' was not declared in this scope
 #define Serial USBSerial
                ^~~~~~~~~
src/config.h:152:26: note: in expansion of macro 'Serial'
  #define dbg_printf(x)  {Serial.printf x;}
                          ^~~~~~
src/ms5611.cpp:97:4: note: in expansion of macro 'dbg_printf'
    dbg_printf(("Tavg : %dC\r\n", temperatureC));
    ^~~~~~~~~~
src/config.h:38:16: note: suggested alternative: 'Serial'
 #define Serial USBSerial
                ^~~~~~~~~
src/config.h:151:26: note: in expansion of macro 'Serial'
  #define dbg_println(x) {Serial.println x;}
                          ^~~~~~
src/nvd.cpp:17:3: note: in expansion of macro 'dbg_println'
   dbg_println(("Preferences 'calib' namespace not found, creating with defaults"));
   ^~~~~~~~~~~
```

SPD13, 2025-03-28:
It worked for me by having the following libs in "lib_deps":
`	ESP32Async/AsyncTCP
	ElegantOTA
	ESP32Async/ESPAsyncWebServer`

har-in-air, 2025-04-05:
Try the latest check in and let me know if you see any issues

har-in-air, 2025-04-08:
Fixed and tested

closed 2025-04-08 by har-in-air