har-in-air/ESP32C3_BLUETOOTH_AUDIO_VARIO #7: How to make it work with the accelerometer z axis not facing in the same direction during calibration and flight?

SPD13, 2025-03-28:
I would like to be able to start the device flat and then hand it to my riser. The orientation during flight will be different from the orientation during calibration as I would like to hand the device on my risers. What modification would you recommend?
Idea 1: Map the calibration process to a button to restart calibration after the device has been powered up. Cons: The paraglider needs to be up and the air and it would require to either let go of the brake of pull the brake toward the riser to press the button
Idea 2: if there is 1g on one axis for X seconds, then use this as the new calibration.
Do you know how "instant varios" are usually managing this problem? I don't see a systematic calibration at each boot up happening on the Flymaster for example.

har-in-air, 2025-03-28:
The accelerometer offset calibration with the vario horizontal only needs to be done once AFTER the board is installed in thecase and before first use.
Gyro offset calibration is done on each power up. For that the only requirement is that the unit should not be disturbed, it can be in any orientation.  If it is disturbed, it will use the previous saved gyro offsetswhich would not be an issue  if you successfully calibrated the gyro in the past few daysand are flying in similar temperature conditions.
After offset calibration the AHRS software uses the computed  Earth coordinates vario orientation to determine the Earth coordinates z axis net acceleration, this is NOT the IMU chip z axis acceleration. 
So the vario does not have to be flat during use, you can mount it to your waist strap or riser.

What can affect the offfset calibrations is if you have removed the board from the case and reinstalled it. The stress from the mounting screws will translate to stress on the MEMS structures in the IMUchip soldered to the board. So if you reinstall the board in the case you should redo the acceleration calibration.
The IMU module should be mounted with double sided foam tape and thin flexible wires to the pcb to mitigate this. . My friend who mounted the modulewith standard 0.1in header pins to the main pcb had some rather large offsets on the accelerometer. I was concerned that he had a defective chip and told himhow I had mounted my board. He reinstalled the module with thin flex wires and the foam tape and had even lower offsets than mine.
Of course the calibration will remove the offsets, but its good to remove external contributing factors.  This is why you see the pcb boards on drones mounted to the chassis with silicone dampers. 

I should make a note of this in the README
For a video demo of vario working in various orientations (post calibration of course) see Kalman Filter Sensor Fusion of Altitude and Acceleration Data 

| 
| 
| 
|  |  |

 |

 |
| 
|  | 
Kalman Filter Sensor Fusion of Altitude and Acceleration Data

 |

 |

 |

SPD13, 2025-03-28:
Thank you so much for the detailed answer! More importantly thank you for this fantastic project!
I am working on a version that uses a cheap esp-32 with a built-in battery charger and I added an e-paper display and a second button, it's working fantastic! I have a clean climb graph and 2 resettable altitude (landing and launching). I tested and have more than 9h battery on a 18650 cell.
I will of course publish the code and documentation on github and give you proper credit.

closed 2025-03-28 by SPD13

har-in-air, 2025-03-28:
Thanks, let me know when you have published the code, I will backport the fixes for the compilation errors. 
regardsHari